machineagency/science-jubilee #11: Sort out if machine is a needed argument to tool base class init

brendenpelkie, 2023-10-10:
Tool base class takes a machine but this doesn't seem to be needed. Is it? I took it out while debugging and never looked into if that will break something or not.

bsubbaraman, 2023-10-11:
It shouldn't be needed; instead, the `_machine` attribute [is set ](https://github.com/machineagency/science_jubilee/blob/ba2617ef22116ae474059c5975bd25c4ad26d831/science_jubilee/Machine.py#L705C29-L705C29)in `load_tool()`. Anecdotally I've been using the repo for a while without passing the machine explicitly and hadn't run into any errors.

I did notice that the Pipette module [takes a machine](https://github.com/machineagency/science_jubilee/blob/ba2617ef22116ae474059c5975bd25c4ad26d831/science_jubilee/tools/Pipette.py#L25) as a parameter, which I _think_ is redundant/not necessary.

MariaPoliti, 2023-11-21:
I've removed the redundant attribute `_machine` from the `Pipette` and `WebCamera` tools. I've kept it in the base `Tool` class, but I am happy to remove it if it is considered uneccessary.