machineagency/science-jubilee #14: Pipette dispense z location specification

brendenpelkie, 2023-10-10:
Add an option and way to specify the z location (as a distance from top or bottom of well) to dispense at. This is important to facilitate mixing or avoid contaminating a tip. Maria might be working on this already so check with here before starting work.

MariaPoliti, 2023-11-13:
This issue has been fixed. It is now possible to specify the location for both `aspirate`  and `dispense` directly by using `locationtype.top(-z)` or `locationtype.bottom(+z)` , where locaiton type should be a `Class Well` type and `z` is the distance in `mm` from bottom or top. 

Note: we might need to add a check for bottom and `-z` values as this might cause the pipette to collide with the bottom of the well. I will add this in in the next few days. Will close this issue when completed.

MariaPoliti, 2023-11-16:
Added check for invalid locations (i.e., below the bottom of the well) and added an option for skipping the check to accomodate sopecific costumization of labware. 

Update is in the `pipette_update` branch. Will close this issue once branch is merged into main.

closed 2023-11-21 by MariaPoliti