machineagency/science-jubilee #18: Discuss deck layout and ordering

brendenpelkie, 2023-10-10:
Current deck layout and ordering (ie where slot #1 is) is backwards from how opentrons does it. Discuss if this is actually how we want it or if we want to stick with convention.

bsubbaraman, 2023-10-11:
Slot 1 is currently set to be closest to Jubilee's origin position. I think having it aligned with Jubilee's axes makes manual adjustments intuitive and some implementation details straightforward (e.g. we can just use the Opentrons labware definitions off-the-shelf), so my instinct is to keep the current layout to encourage people to 'buy in' to the Jubilee coordinate system. This could be beneficial to build a sound mental model of the machine should users want to make their own non-deck related applications in the future. But happy to change it if others feel that it is unnecessarily confusing for scientists already comfortable with OpenTrons!

bsubbaraman, 2023-11-08:
Closing this-- feel free to reopen if we should discuss more!

closed 2023-11-08 by bsubbaraman