machineagency/science-jubilee #33: No safe_z if a deck isn't configured

bsubbaraman, 2023-11-29:
[`requires_safe_z` ](https://github.com/machineagency/science_jubilee/blob/6ad597817d5dd5ddc346329ac43f0f4fd07d02ca/science_jubilee/Machine.py#L82) sets its value from the Deck, but this is an issue if you don't have a deck configured yet (or you want to use science_jubilee without a deck). 

I propose adding a check to see if there is a Deck installed, and if not, set a default value + raise a warning for the user to manually set a safe_z if they have objects on the platform.

bsubbaraman, 2023-11-29:
I've made this change and can add it in my next commit!

brendenpelkie, 2023-12-08:
Didn't see this issue filed here but did run into this problem with some recent testing. I just changed the default value on deck setup load from 'None' to 10mm, and pushed this in #40 . Sounds like your solution might be more elegant.

PR #58 (2024-02-28): Safe z fix

Resolves #33

 Changes:

- Added a default safe_z value to address #33 
- Added new labware defintion for 6 well palte

#### PR Checklist

- [x] [Inline documentation] is included / updated
closed 2024-02-28 by bsubbaraman