machineagency/science-jubilee #42: Well plate alignment tool

brendenpelkie, 2023-12-08:
### Feature request details

Add a software tool workflow to set wellplate offsets and alignment. This would look something like it does  on opentrons. Here, the user is guided through a series of steps to place the tool tip in a reference location over the well plate, then use this to update well plate locations. Ideally this would apply a location correction as well.

Sonication station supports this and I implemented this on my color mixing demo so the logic is written, this is a matter of integrating this into the new library.

bsubbaraman, 2023-12-11:
Just to clarify, is this meant as a way to create a interactively create a new labware definition instead of using [the labware designer ](https://labware.opentrons.com/create/), or something else? If there's a link to the relevant OpenTrons documentation that might be helpful!

brendenpelkie, 2023-12-11:
This is to fine-tune wellplate alignment after you have it loaded, before you run an experiment. Even on the OT2 deck there is always some variation in positioning, so it is useful to have a method to account for that. Its mentioned in the OT2 docs here but haven't found anything more detailed: [https://docs.opentrons.com/v2/robot_position.html#using-labware-position-check](url).

OT2 doesn't do a rotation correction, while the original sonication station position check does do a rotation correction by having the user sample 3 points then doing some trigonometry. I think this is worth having given the loose tolerances of our current deck designs.

bsubbaraman, 2023-12-11:
Gotcha! That seems great, especially for using 96-well plates and beyond