machineagency/science-jubilee #100: Docs are broken

brendenpelkie, 2024-05-12:
### Steps to reproduce this bug

### Steps:
1. Click link to documentation
2. Be sad there is no documentation
3.
....

### Snippet:

```python

// Paste your code here :)

```

bsubbaraman, 2024-05-12:
I think this is from due to @sgbaird's ongoing updates and change to readthedocs (https://github.com/sgbaird/science-jubilee)! There are docs here https://science-jubilee.readthedocs.io though the formatting is funny right now

sgbaird, 2024-05-13:
With the change from `science_jubilee` to `science-jubilee` that I suggested, the GitHub pages site went from https://machineagency.github.io/science_jubilee to https://machine agency.github.io/science-jubilee. Like @bsubbaraman mentioned, the plan is to shift to readthedocs. The PR will need to be merged. I will address some of this today.

bsubbaraman, 2024-05-13:
ah of course, I forgot about the name change! Thanks, Sterling. I've updated the readme link and closed this for now, with more readthedocs changes coming soon!

PR #101 (2024-05-13): update docs website to science-jubilee

Resolves #100 

 Changes:

We renamed the repository `science-jubilee` from `science_jubilee`, so updating the github pages url in the readme accordingly.
closed 2024-05-13 by bsubbaraman