machineagency/science-jubilee #116: Installation on Raspberry Pi 5

sgbaird, 2024-05-28:
### Steps to reproduce this bug

Reported by @yakavetsiv
```
      -- Installing: /tmp/pip-install-vzi0mxu1/opencv-contrib-python_174a793d581f4484b688df56df0872c8/_skbuild/linux-aarch64-3.11/cmake-install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
      Copying files from CMake output
      Traceback (most recent call last):
        File "/home/ac-duet/science-jubilee/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/ac-duet/science-jubilee/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/ac-duet/science-jubilee/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-rxe8vw61/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 230, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-rxe8vw61/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-build-env-rxe8vw61/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 268, in run_setup
          self).run_setup(setup_script=setup_script)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-rxe8vw61/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 537, in <module>
          main()
        File "setup.py", line 267, in main
          setup(
        File "/tmp/pip-build-env-rxe8vw61/overlay/lib/python3.11/site-packages/skbuild/setuptools_wrap.py", line 706, in setup
          _classify_installed_files(
        File "setup.py", line 411, in _classify_installed_files_override
          with open(config_py, 'w') as opencv_init_config:
               ^^^^^^^^^^^^^^^^^^^^
      FileNotFoundError: [Errno 2] No such file or directory: '_skbuild/linux-aarch64-3.11/cmake-install/python/cv2/config-3.py'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for opencv-contrib-python
  Building wheel for pyzmq (pyproject.toml) ... done
  Created wheel for pyzmq: filename=pyzmq-26.0.3-cp311-cp311-linux_armv8l.whl size=707528 sha256=d17b9efc6687e83c0339d6eba3083ef72a25bd3490b1a88445c77b254016a982
  Stored in directory: /home/ac-duet/.cache/pip/wheels/f4/44/89/690cd7f3816cdecb2a681775e2fc6f23c25bd0005125ea29ba
Successfully built science-jubilee pyzmq
Failed to build opencv-contrib-python
ERROR: Could not build wheels for opencv-contrib-python, which is required to install pyproject.toml-based projects
```

sgbaird, 2024-05-30:
In my haste, posted separately at https://github.com/machineagency/science-jubilee/issues/121, but moving to here to consolidate.

Reported by @yakavetsiv when trying to separate out `science-jubilee` from Open CV dependency, using Raspberry Pi 5 with bookworm Duet firmware, as a follow-up to https://github.com/machineagency/science-jubilee/issues/116.

Splitting dependencies is currently underway #120. @yakevitsiv I suggest switching to the `deps` branch and using that to debug. Planning to merge once I've verified installation is OK.

```python
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/core/__init__.py:24
     23 try:
---> 24     from . import multiarray
     25 except ImportError as exc:

File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/core/multiarray.py:10
      9 import functools
---> 10 from . import overrides
     11 from . import _multiarray_umath

File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/core/overrides.py:8
      7 from .._utils._inspect import getargspec
----> 8 from numpy.core._multiarray_umath import (
      9     add_docstring,  _get_implementing_args, _ArrayFunctionDispatcher)
     12 ARRAY_FUNCTIONS = set()

ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/__init__.py:130
    129 try:
--> 130     from numpy.__config__ import show as show_config
    131 except ImportError as e:

File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/__config__.py:4
      3 from enum import Enum
----> 4 from numpy.core._multiarray_umath import (
      5     __cpu_features__,
      6     __cpu_baseline__,
      7     __cpu_dispatch__,
      8 )
     10 __all__ = ["show"]

File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/core/__init__.py:50
     27     msg = """
     28 
     29 IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
   (...)
     48 """ % (sys.version_info[0], sys.version_info[1], sys.executable,
     49         __version__, exc)
---> 50     raise ImportError(msg)
     51 finally:

ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/ac-duet/science-jubilee/.venv/bin/python3"
  * The NumPy version is: "1.26.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

ImportError                               Traceback (most recent call last)
Cell In[1], line 1
----> 1 from science_jubilee.Machine import Machine
      3 m = Machine()                 # a new machine, called 'm'
      4 m.move_to(x=100, y=50, z=200) # moves to the absolute position (100, 50, 200)

File ~/science-jubilee/src/science_jubilee/Machine.py:16
     13 import requests  # for issuing commands
     14 from requests.adapters import HTTPAdapter, Retry
---> 16 from science_jubilee.decks.Deck import Deck
     17 from science_jubilee.tools.Tool import Tool
     19 # TODO: Figure out how to print error messages from the Duet.
     20 
     21 
     22 # copied from machine agency version, may not be needed here

File ~/science-jubilee/src/science_jubilee/decks/Deck.py:6
      3 from dataclasses import dataclass
      4 from typing import Dict, Tuple
----> 6 from science_jubilee.labware.Labware import Labware
      9 @dataclass
     10 class Slot:
     11     """Class that defines a slot on the Jubilee deck.
     12     Each slot has a unique index, offset, and can have a labware associated with it.
     13 
   (...)
     21     :type labware: :class:`Labware`
     22     """

File ~/science-jubilee/src/science_jubilee/labware/Labware.py:9
      6 from math import acos, cos, sin, sqrt
      7 from typing import Dict, Iterable, List, NamedTuple, Tuple, Union
----> 9 import numpy as np
     12 @dataclass
     13 class Well:
     14     """A class representing a well of a labware.
     15 
     16     Each Well is associated with a specific name, depth, total liquid volume, shape, diameter,
   (...)
     20     :rtype: :class:`Well`
     21     """

File ~/science-jubilee/.venv/lib/python3.11/site-packages/numpy/__init__.py:135
    131 except ImportError as e:
    132     msg = """Error importing numpy: you should not try to import numpy from
    133     its source directory; please exit the numpy source tree, and relaunch
    134     your python interpreter from there."""
--> 135     raise ImportError(msg) from e
    137 __all__ = [
    138     'exceptions', 'ModuleDeprecationWarning', 'VisibleDeprecationWarning',
    139     'ComplexWarning', 'TooHardError', 'AxisError']
    141 # mapping of {name: (value, deprecation_msg)}

ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
[ ]:
```

Comments from @bsubbaraman:
- Bookworm was released recently, so OpenCV for that version hasn't been debugged/integrated yet

Related #94

yakavetsiv, 2024-05-30:
The following solution for Issue #121 was found : `sudo apt-get install libopenblas-dev`

closed 2024-06-14 by sgbaird