machineagency/science-jubilee #137: Duel 3 Electronic Backboard Assembly

ethraj2001, 2024-07-02:
### Feature request details

Hi there, 

I was just wondering where I can find the instructions for the duet 3 electronics panel. I cannot seem to find it on the wiki. I only see the duet 2 electronics panel. The back panels are a little confusing so any help would be appreciated. 

[duet2_electronics_panel_assembly.pdf](https://github.com/user-attachments/files/16073266/duet2_electronics_panel_assembly.pdf)

Regards
Ethan R.

sgbaird, 2024-07-02:
The documentation on that needs to be improved. For at least the wiring, there is https://github.com/machineagency/science-jubilee/blob/main/docs/pdfs/jubilee_wiring_colormix.pdf and https://raw.githubusercontent.com/machineagency/jubilee/main/frame/assembly_instructions/wiring/duet3_mini/duet3_mini_frame_wiring.png.

There is [a picture of `reefingninja`'s setup from the Jubilee discord](https://discord.com/channels/627982902738681876/627988602269335568/1182078408084172871) (not Science Jubilee discord):

[image]

ethraj2001, 2024-07-02:
okay I see a wiring and layout but I am not sure what screws to use etc.