machineagency/science-jubilee #146: OT-2 pipette tool

yakavetsiv, 2024-07-26:
### Feature enhancement details

The current models of the 3D printed OT-2 adapter have multiple issues:

1. Tool Post Width: The tool post is wider than the adapter, causing misalignment.
2. Base for Wedge Plate: The base is too deep, preventing it from locking properly.
3. Wings and Wedge Plate: The wings are too low, and as a result, the wedge plate hole is also too low, which prevents proper locking.
4. Plastic Springs: The plastic springs tend to compress excessively and require reinforcement.
5. Metal Plate Requirement: There is a need for a metal plate to improve structural integrity.
6. Shift of the pipette from a perpendicular alignment over time.
7. Absence of model for wire holder

Attached are the *.STL models of the updated OT-2 holder for your review.

[OT-2 printing.zip](https://github.com/user-attachments/files/16385597/OT-2.printing.zip)

brendenpelkie, 2024-08-21:
Thanks for these suggestions @yakavetsiv. 

1. Have you put your re-designed parts into use? Have they resolved any of these issues?
2. Yes the tool post published with the pipette tool is the wrong size for the 3D printed version. That needs to be updated. 
3. What do you mean by 'metal plate requirement'? The currently published 3D printed flexure mount uses a machined tool changer plate. (https://github.com/machineagency/science-jubilee/blob/main/tool_library/OT2_pipette/assembly_docs/OT2_Pipette_3D_only_assembly_instructions.pdf). Is this what you meant, or are you referring to something else?
4. How is the 'bracket-P200' part in your stl files assembled? Does that clip around the pipette body somehow? Is this part for vertical alignment?

If you've put your redesign into use and think it is an improvement over the original, it would be great to incorporate your version as the new 3D printed option.

yakavetsiv, 2024-08-22:
1. Yes, I'm currently using multiple OT-2 pipette tools with the designed parts, and the updated components have resolved all the previously mentioned issues.
2. I used the published tool post as a reference.
3. Apologies for the incorrect terminology—I was referring to a machined tool changer plate. The updated design no longer requires this part; only the metal balls and Delrin insert are needed for the locking mechanism.
4. I’ll send you photos of the assembly tomorrow. It is indeed clipping around the pipette body, leaving only the vertical movement unconstrained.

PS. I also filmed the wiring of the OT-2 tool. Will be happy to shape it for WIKI

brendenpelkie, 2024-08-22:
Glad to hear you got it working. It sounds like this design improves on our currently published version. If you are OK with it, I can replace the current 3D printed version design with your updated one. Would you be able and willing to share your CAD files in addition to the STLs?

A wiring video would be a great addition to our pipette tool documentation!