machineagency/science-jubilee #151: Updating travel docs

sgbaird, 2024-07-28:
For example, Science Jubilee dimensions, 10" minimum height if using short extrusions

brendenpelkie, 2024-08-20:
@sgbaird could you add a bullet point to the "Packing and Transport" section of the jubilee travel docs mentioning how your packing strategy for Accelerate worked out? I think that will be a useful anecdote for future Jubilee travel.

sgbaird, 2024-08-20:
Summarizing here and then will port over a trimmed version:

300' of bubble wrap, cardboard slats, packing wrap, and two patio furniture bags

[image]

[image]

Having to unwrap everything at the airport due to the bag not fitting through the oversized scanner at YYZ. During re-packing, one of belts came loose. Can be fixed.

[image]

Then, it not coming on the same plane and it coming too late for the actual demo.

Return trip had almost no issues (fit through the scanner, picked it up).

brendenpelkie, 2024-08-21:
Awesome thanks! Did it survive OK? Any damage? Was the alignment reasonable after it made it home?