machineagency/science-jubilee #154: Heat Set pieces and Force

owen-melville, 2024-08-15:
Overall issue: When I tried to tighten the pieces in the "Motor Installation" section of the "Toolchanger Remote Elastic Lock Assembly" the heat-set brass inserts came out, and the piece was not held in.
[image]
[image]

Suggestions:
-The 3D printed part (RLB-11) seems to have too much space for the inserts, which doesn't allow the insert to be as stiffly locked in place as it could be
-In the instuctions, make a note about not over-tightening
-Using superglue on the inserts?

owen-melville, 2024-08-15:
I tried again with new piece... exact same issue... I am proceeding but it does not seem very secure and I can see the heart set insets coming out

owen-melville, 2024-08-15:
I think potentially this piece should have a different default orientation (large side down) for printing

owen-melville, 2024-08-15:
to be honest I have seen lots of the heat set inserts start to pull out as I'm assembling

bsubbaraman, 2024-08-15:
Hey @owen-melville thanks for the comments! Checking a couple things off the bat, since you said this is happening on multiple parts: did you adjust your slicer settings to use 6 perimeter layers? (This is important for parts with heat set inserts, along with the other default print settings listed here https://jubilee3d.com/index.php?title=3D_Printed_Parts).  And if you measure the interior diameter of your printed parts, is it ~5.1mm?

owen-melville, 2024-08-15:
Yes - using Bambu studio on advanced I set "Wall Loops" to 6 and infill to 20% but "Outer Perimeter" layer first was not a clear option. I didn't measure but the holes did seem a bit large, there was some resistance to push the piece in with the soldering iron but it didn't seem like much... I am wondering why this piece has the default orientation that it does.... When you put it "Large" size down, the holes are printed vertically rather than horizontally. The only holes that are switched to horizontal are the small holes that I needed to use a drill to do pilot holes on anyway.

owen-melville, 2024-08-15:
It is caused by the holes being printed horizontally...  See top of the two. Has this come up before? 
[image]

bsubbaraman, 2024-08-16:
Thanks for the extra tests & pics! I haven't run into this personally, but print orientation can definitely effect dimensional accuracy. There might be some printer calibration routines to mitigate the effects--holes coming out as ovals is a problem that is documented here https://forum.bambulab.com/t/why-ovals-and-not-circles/7354 on Bambu printers and the thread might have possible fixes (assuming you're using a Bambu printer, since you mentioned Bambu studio earlier).

In general the print orientations are chosen to maximize mechanical strength/minimize supports. In this case though, I think an print orientation might be a good fix!

owen-melville, 2024-08-16:
Yeah - Our colleague printed on another Bambu printer (different one, different building) and apparently had the same result. I'll take a look at that thread and thanks for sharing!

yakavetsiv, 2024-08-17:
In addition to the printer settings, another potential issue could be the outer diameter of the heat inserts. The kit includes smaller heat inserts (shown on the right in the photo), which are too small for the intended holes. Using heat inserts with a larger outer diameter (shown on the left in the photo) resolves this problem.
[image]

brendenpelkie, 2024-08-20:
For future procurement needs, McMaster Carr has the correct sized 5.1mm OD inserts as product number 94180A331. (https://www.mcmaster.com/94180A331/). Filastruder also used to stock these but I'm not seeing them on their website currently.