machineagency/science-jubilee #185: Driver Failure

jzhang-03, 2025-02-03:
The U-Axis of the tool holder suddenly stopped moving for me. I received an error message that states: **"Driver 0.1 warning: phase A may be disconnected, phase B may be disconnected."** Driver 0.1 is the driver for moving the motor that controls the U-Axis of the tool holder. However, movement in the X, Y, and Z axis are all fine and result in no errors. The machine wasn't particularly hot either.

This error occurred after trying to home all the axes whilst accidentally having the gel extruder tool selected (but not actually loaded onto the tool holder). This resulted in the tool holder physically crashing into the frame of the jubilee. I didn't notice that I had the gel extruder tool selected, so I tried to home the machine a few times, which all resulted in the tool holder crashing into the machine. Once I had deselected the gel extruder, I was able to properly home the X, Y, and Z axis, however the U-Axis on the tool holder was no longer moving and the error message mentioned above appeared. The error message seems to only appear when I try to the move the U-Axis in increments of 50 or more, but doesn't appear when I try to move the U-Axis in increments of 10 or less.

I was wondering if there are some guidelines that I can use to troubleshoot and diagnose the source of the problem.

The Jubilee that I am operating on uses RepRapFirmware for Duet 3 Mini 5+ version 3.5.3 (2024-09-18) and is running on Duet 3 Mini5plus WiFi (SBC mode).

jzhang-03, 2025-02-03:
Motor was disconnected.

bsubbaraman, 2025-02-03:
Glad it was figured it out---thanks for posting, could be useful for others to see this error + fix on here!