machineagency/science-jubilee #199: Alternate to seeed?

bobthechemist, 2025-04-15:
### Feature enhancement details

One issue that came up with the AS7341 spectral sensor is how it is dependent on the microcontroller being SEEED based. (src/science_jubilee/tools/AS7341.py) Since CircuitPython is used for the microcontroller, a more generalized approach might be to use the [Adafruit board toolkit](https://docs.circuitpython.org/projects/board-toolkit/en/latest/index.html) which returns any CP-based board.

danlil666, 2025-04-15:
good suggestion! will try to implement this.

jschrier, 2025-04-18:
@bobthechemist : It would be neat to make a notional implementation with a [little RP 2040 QT board](https://www.adafruit.com/product/4900), as all this tool really needs is a STEMMA and USB ports, and it would fit more nicely into the tool package.