machineagency/science-jubilee #212: Pipette tool doesn't attempt to pick up a tip when calling pickup_tip()

ststewa3, 2025-08-26:
### Duet3D Firmware Version

FIRMWARE_NAME: RepRapFirmware for Duet 3 Mini 5+ FIRMWARE_VERSION: 3.5.4 ELECTRONICS: Duet 3 Mini5plus Ethernet FIRMWARE_DATE: 2024-11-24 10:44:24

### Steps to reproduce this bug

### Steps:
1. Turn on and home Jubilee (including V axis)
2. Run the code snippet below
3. Pipette moves to the correct XY position, but then moves the bed down instead of up when attempting to pick up a pipette tip

### Snippet:

```python

from science_jubilee.Machine import Machine
from science_jubilee.tools.Pipette import Pipette
from science_jubilee.labware.Labware import Labware

m = Machine(address="10.155.55.214")
m.home_all()

deck = m.load_deck("lab_automation_deck_CALIBRATION_4")

# Now install labware in correct positions
m.move_to(z=150)
tiprack = m.load_labware("opentrons_96_filtertiprack_1000ul", 0)
trash = m.load_labware("generic_petri_dish_100ml", 1)
samples = m.load_labware("nest_96_wellplate_200ul_flat", 2)
stocks = m.load_labware("nest_12_reservoir_15ml", 3)

pipette_index = 1
pipette_config = "P1000_config.json"
pipette_name = "Pipette"

pipette = Pipette.from_config(pipette_index, pipette_name, pipette_config)
m.load_tool(pipette)

pipette.add_tiprack(tiprack)
pipette.trash = trash[0]

m.pickup_tool(pipette)

pipette.pickup_tip()

# Pipette moves to correct XY position, but moves the bed down instead of up to pick up a tip

```

### Notes:

- I double checked my wiring from [this tutorial](https://science-jubilee.readthedocs.io/en/latest/building/pipette_tool.html) and everything matches up
- My config.g file also follows the tutorial. I copied and pasted it at the end of this issue in case it is useful to look at.
- Pipette.py API code also matches what's currently on the Science Jubilee GitHub
- Video of the issue:

https://github.com/user-attachments/assets/55d30ca5-6329-4ffb-947f-e749bfa98751

### My config.g file:

```
; Default config.g template for DuetPi
; Replace this with a proper configuration file (e.g from https://configtool.reprapfirmware.org)

; Display initial welcome message
M291 P"You are operating an SBC image in standalone mode. Please see <a href=""https://www.duet3d.com/StartHere"" target=""_blank"">here</a> further information." R"Welcome to your new Duet 3!" S1 T0

; Enable PanelDue
M575 P1 B57600 S1

[... 129 lines cut ...]
M208 U0:200            ; Set Elastic Lock (U axis) max rotation angle

; Bed Heater and Temperature Sensor
;-------------------------------------------------------------------------------
; Define Built-in Thermistor Settings
M308 S0 P"temp0" Y"thermistor" T100000 B3950 A"Bed" ; built-in Keenovo thermistor
; Define Heater 0
M950 H0 C"out0" T0      ; H = Heater 0
                        ; C = heater output pin
                        ; T = assigned temperature sensor
M143 H0 S130            ; Set max bed temperature to 130C    
M140 H0                 ; Assign Heater 0 to the bed

; This config requires a heater-tuning procedure to produce a valid M307 command.

; Tool Definitions
;-------------------------------------------------------------------------------
M563 P0 S"Camera"          ; Camera with tool index 0
M950 P0 C"0.out5"          ; Camera LED lights with LED index 0
M563 P1 S"P1000 Pipette"   ; Pipette with tool index 1
M563 P2 S"Spectrometer"    ; Spectrometer with tool index 2

M98  P"/sys/toffsets.g" ; Load tool offsets from the Control Point from ext file.
M501                    ; Load saved parameters from config-override.g
```

bsubbaraman, 2025-08-26:
Hi @ststewa3! It looks to me like the system thinks the endstop is already engaged when starting the pickup, and is moving immediately to a `safe_z` position. Is the pipette pickup endstop working as expected? If you check the endstop status (e.g. with an `M119` commands sent from the duet web console), is the Z endstop disabled before pickup? And if you manually engage the pipette endstop and hold it while sending the `M119` command again, is it enabled?

Let me know what you find, or I can follow up with more info!

ststewa3, 2025-08-26:
Hi @bsubbaraman, it says that the V axis is not stopped when the limit switch is pressed and not pressed. I also checked the switch with a multimeter and there is no difference in resistance when the switch is pressed or not pressed. I guess there's just a bad connection so I'll replace the switch and see if that helps.

bsubbaraman, 2025-09-27:
Great, thanks for letting me know how it went! Happy pipetting 🧪